upb-lea/openmodelica-microgrid-gym #11: more examples

stheid, 2020-05-05:
-  first split examples.rst into subfiles for each example
- add the two simple examples to the userguide
```python
import gym
import numpy as np

if __name__ == '__main__':
    env = gym.make('openmodelica_microgrid_gym:ModelicaEnv-v1',
                   model_input=['i1p1', 'i1p2', 'i1p3'],
                   model_output=dict(lc1=['inductor1.i', 'inductor2.i', 'inductor3.i']),
                   model_path='grid.network.fmu')

    env.reset()
    for _ in range(1000):
        env.render()
        env.step(np.random.random(3))  # take a random action
    env.close()
```
```python
import gym
import numpy as np
import pandas as pd
from openmodelica_microgrid_gym import Agent, Runner

inputs = ['i1p1', 'i1p2', 'i1p3']

class RndAgent(Agent):
    def act(self, obs: pd.Series) -> np.ndarray:
        return np.random.random(len(inputs))

if __name__ == '__main__':
    env = gym.make('openmodelica_microgrid_gym:ModelicaEnv-v1',
                   model_input=inputs,
                   model_output=dict(lc1=['inductor1.i', 'inductor2.i', 'inductor3.i']),
                   model_path='grid.network.fmu')

    agent = RndAgent()
    runner = Runner(agent, env)

    runner.run(1)
```

stheid, 2020-05-05:
* Add very simple "out-of-the-box" examples including code snippets into docu (plus readme)
* Add selected code parts (e.g. controller setup) into example docu

h-bode, 2020-05-05:
I assign it to myself, if it´s okay for you :-)

stheid, 2020-05-05:
Maybe it makes sense to rather put the the code into the getting started file in the userguide.

if found looking at the Miniconda documentation quite inspiring. they seem to have a good structure.

closed 2020-05-11 by h-bode