upb-lea/openmodelica-microgrid-gym #13: improve example integration and dependency management

stheid, 2020-05-05:
currently installing `pip install openmodelica-microgrid-gym' will not provide the complete installation needed to execute the env, as openai gym will import everything, including the saveoptagent and therefore failing

it will only work if `pip install openmodelica-microgrid-gym['examples']` is installed. however, this is suboptimal, as this should only be needed if one actually wants to run the examples.

* [ ]  maybe we need to find a way not to package the examples and also exclude the examples related agents if they are not installed, removing those strong dependencies.

---
currently we have:
*  requirements.txt
*  requirements-dev.txt
*  setup.py has
    * setup require
    * test require
    * extra require

stheid, 2020-06-12:
I think here we could get some inspiration from tensorforce once again.

There travis and all developers should install all full dependencies. however, if imports are missing its not an issue if the classes and modules containing the missing deps are never called.

So it will be mainly an documentation issue to make it obvious to the user what example needs which dependencies.

Webbah, 2020-12-18:
As discussed with @stheid no longer relevant/ unlikely to be solved

closed 2020-12-18 by Webbah