upb-lea/openmodelica-microgrid-gym #7: Best parameter set output after termination of SafeOpt agent

wallscheid, 2020-05-05:
In any SafeOpt or similar example it would be helpful:
*  to get a final message on the best found parameter set including its performance value 
*  a re-plot in the time domain (current, voltages, etc.) of the system behavior using this best parameter set
*  a re-plot of the parameter space highlighting the best found parameter set (e.g. by encirling this point using another color)

If more than 10 SafeOpt episodes are run, I find it difficult to search for the optimum parameter point in the command line result message as well as in the plots window manually. Hence, I believe a report on the found optimum as suggested above would be helpful especially for an increasing number of optimization iterations. The command line report on the optimum should be clearly distinguishable from the command line outputs during the optimization (e.g. using separation lines or similar).

Webbah, 2020-05-14:
a re-plot of the parameter space highlighting the best found parameter set (e.g. by encirling this point using another color)
will eb done in #25

Webbah, 2020-05-14:
- to get a final message on the best found parameter set including its performance value

Done

Webbah, 2020-05-14:
- a re-plot in the time domain (current, voltages, etc.) of the system behavior using this best parameter set

Where to plot (runner, agent, env, main)?
Only currents and voltages? What to plot is pre-defined in env.

Can be executed in the end using best control parameters for the agent and run the runner with that parameters again.
Problem: Agent gets reseted in the runner, parameter are overwritten

Webbah, 2020-05-25:
Done, output as follows:

           J                                      Params
0 -0.527522                                [0.01, 10.0]
1 -0.450459   [0.01517286546392185, 14.272587867479627]
2 -0.315363  [0.014080419150519476, 47.177210473910996]
3 -0.283602  [0.017012027162882904, 101.12828593544891]
4 -0.306049  [0.0028663300205910283, 81.51673234870911]
5 -0.385075                   [0.0, 18.804784877211315]
6 -0.699400    [0.02495826942674722, 52.62289236271889]
7 -0.285122  [0.007083126598803516, 113.19739691891589]
8 -1.335885  [0.015830994628372473, 170.18604221050606]
9 -0.885461   [0.022891364186758124, 84.81356763505792]

 Experiment finished with best set: 

  Kpi = [0.017012027162882904, 101.12828593544891]
  Resulting in a performance of J = -0.2836023386749026

Best experiment results are plotted in the following:

[image]
[image]

Webbah, 2020-05-25:
Setpoints and flexible measurements plots shifted to #12

closed 2020-05-25 by Webbah