upb-lea/openmodelica-microgrid-gym #19: parallelisation of steps

h-bode, 2020-05-08:
Investigate the possibilities of process parallelisation for simulation time optimization and the possibility of simulate larger system

h-bode, 2020-05-11:
https://info.ornl.gov/sites/publications/files/Pub62531.pdf

might have some interesting ideas. 

splitting the system in several subsystems is completely thrown away?

h-bode, 2020-05-12:
https://openmodelica.org/images/M_images/OpenModelicaWorkshop-2016/OpenModelica2016-talk11-Martinflehmig-Design%20Approach%20for%20a%20Scalable.pdf

Really good overview of current state of the art and problems. As far as I know, it is not possible to run one single FMU on multiple cores. Coupled FMU´s might be a way. The guys in the fist paper reduced the simulation time with perfect conditions on a tuned data set by about 40%. Since the simulation time is at the moment not the most time consumig point, I am not sure if making a multi FMU problem was the best way to speed up the toolbox

h-bode, 2020-05-16:
Access to the gitlab of the "Parallel Execution of FunctionalMock-up Unitsin Buildings Modeling" comparision granted. (https://info.ornl.gov/sites/publications/files/Pub62531.pdf) 

 No code inside available, asking the author for the reason. 

Last Mail: I gave access to the repo. You should have received the invitation. So, as far as I remember, i basically wrote a wrapper in C++ to use the FMU functions (model exchange FMU). Then coupled them as described in the document and solved with a simple Euler solver – implemented in C++ again. I hope it would be beneficial for your purposes. 

No usable interface for Python. Think I will get access on monday, author was really willing to help in previous mails. Discussion on monday if it was possible to include a whole c++ wrapper, maybe Stefan can say something about this?

stheid, 2020-05-18:
Python can call native C++ code. Numpy does that for example. I have no idea on the details.

closed 2020-08-07 by h-bode