upb-lea/openmodelica-microgrid-gym #26: Built active load

Webbah, 2020-05-15:


h-bode, 2020-05-25:
concept of active load works. 

2 params need to be transferred to the FMU
- the setpoint of the voltage (not needed if default of 230V * sqrt(2) is used)
- the power consumption of the load in Watt (default 5 kW, approx. 10.5 Ohm resistor)

Picture compares a resistor with 10.58 Ohm (phase 2, blue) with an active load, consuming 5 kW (phase 3, green)

h-bode, 2020-05-25:
[image]

h-bode, 2020-05-25:
Example pushed to new feature brand 26

h-bode, 2020-06-15:
From now, the nominal values of frequency and voltage are used to calculate the resulting resistances or inductivities/capacities to create loads of the demanded power. 

Issue #40 shows a way to include a PI controller which takes care of changes in the grid voltage, but does not consider changes in the frequency.

h-bode, 2020-06-15:
In FMU´s, calculation of parameters are only happening ones at the beginning. If q_ref and p_ref are set in OpenModelica, the correct values for resistors/inductors/capacitors will be calculated. 

Changing this parameters in python needs a small workaround, which is shown in the picture below:

[image]

the values for p_ref and q_ref can be set above together with the general grid parameters like v_DC.

In Openmodelica, an automatic decision between a positive and a negative reactive load is possible by setting q_ref either to a positive or negative values. Since in python the specific parameter of the state equation needs to be set, it has to be predefined if the reactive load is positive (use 'serial_load.reactive_load1.L': p.reactive_pos) or negative (use 'serial_load.reactive_load1.C': p.reactive_neg).

h-bode, 2020-06-15:
Since everything which is included here is also included in the feature branch #40 which has newer versions of the rest of the code, it is not recommended to use this branch. This features will come together with #40 to the delevop.

closed 2020-08-07 by h-bode