upb-lea/openmodelica-microgrid-gym #38: Turn off agent and/or env render

Webbah, 2020-06-03:
Env-render: save list (idx) steps-wise if close = False else: plot list
Agent-render: self.viz = True => render

h-bode, 2020-08-28:
Right now, in the develop, there is an error while running single_inverter_voltage_current_control.py.

to avoid this error and make the program run, I commented in the runner the lines 62+63

          #  if visualise:
           #     agent_fig = self.agent.render()

@stheid , okay to use it liek this in the pull request? Some of the plots are missing, but at least, it does run.

stheid, 2020-08-28:
i think this is not really related to the issue. The problems you mention should be solved within the agent render function of the safeopt agent and are only caused by trying to plot 4d plots.

stheid, 2020-10-09:
Current behaviour:
- env.render does basically nothing
- env.close does the final rendering after episode

needed behavour
- env.close is called always and not-calling render will correctly effect the behaviour in close (hence not plot anything in close when render has not been called during the episode)

-> env.render needs to store information, about the steps that should be plotted and than calling close will render all of the selected steps. Therefore, not calling render will also result in no plot when close is called. Therefore the modification in the runner is simple: viz=false -> do not call render

stheid, 2020-10-09:
Alternative B:
substitute current env.render implementation completely with tensorboard summary recorder functions.

advantage against the previous sugestion:
- tensorboard continuos plotting and monitoring, expecially cool for longer runns

disadvantage:
- no popping up plots
- ditching all matplotlib code (which already has quite some functionality)

stheid, 2020-10-09:
we decided for the first option and keep the idea with tensorboard for later:
- the env.render fucntion will set a "do-render" flag true and this will be read when we are calling close
- additionally the env.reset function will reset this flag to false.

we do not store any information about how and when the rendering will be called as it only makes sense to render the whole plot continuosly

closed 2020-10-15 by stheid