upb-lea/openmodelica-microgrid-gym #39: Remove unused qdroop argument

stheid, 2020-06-12:
https://github.com/upb-lea/openmodelica-microgrid-gym/blob/master/examples/single_inverter_current_control_safe_opt.py#L159-L165

The argument is not used inside the controller. Is this a functional or usability bug?

Webbah, 2020-07-27:
@stheid : it is not used due to the controller inherits from the VtgController, which has a qDroop

Task of this controller is to supply the current setpoint nevertheless what the grid is doing, so no droop.
The pDroop ctrl is used to track the frequency needed for the PLL...

That is maybe a problem....

This controller should not dropt the frequency due to loading.
So we would need a different class this controller can inherit from or write it from scratch.
So no pDroop and no qDroop needed. But PLL and that frequency has to be measured. 

So it should inherit from the current controller!

Problem is, that in our case the grid consists only of this inverter.

Webbah, 2020-10-09:
Solution:
- Add a different parent class the currentSourcing inverter can inherit from
- New class should be between Controller <-> Voltage/CurrentController Class
1. Controller
2. XXX adding DDS
3. Voltage/CurrentController adding Droop

-> Droop parameters can be deleted then
https://github.com/upb-lea/openmodelica-microgrid-gym/blob/master/examples/single_inverter_current_control_safe_opt.py#L159-L165

closed 2020-10-12 by stheid