upb-lea/openmodelica-microgrid-gym #44: Bound action space

stheid, 2020-06-17:
In the past i had several problems with RL algorihms and libraries related our environments action space being unbounded.

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/00e28db1fa0ab6350db413292a4ea10f626ee32c/openmodelica_microgrid_gym/env/modelica.py#L165

i propose to think of reasonable action values and restrict the allowed actions much stronger.

https://github.com/tensorforce/tensorforce/issues/696
https://github.com/hill-a/stable-baselines/issues/895

stheid, 2020-06-17:
https://stable-baselines.readthedocs.io/en/master/guide/rl_tips.html#tips-and-tricks-when-creating-a-custom-environment

The actions should actually be normalized to [-1,1]

wallscheid, 2020-06-17:
Yes, normalization would be a good thing. Both for actions and states.

closed 2020-07-24 by stheid