upb-lea/openmodelica-microgrid-gym #45: Rework of Env

stheid, 2020-06-26:
- [ ] Normalize actions #44 
- [ ] Normalize states
- [ ] Update standard agents to be capable of denormalization
- [ ] All values in ABC
- [ ] Reference in the Environent #43 

The referenced issues showed that a general overhaul of the environment interface is needed in order to allow general RL agents to learn on it properly

The new environment will provide normalized states including reference information. Everything voltage and current will be in ABC.
The actions expected by the environment are normalized as well.

It remains to be seen if an stateless RL agent can act well on ABC data, but for now it seems like a good path. Maybe it makes sense to keep the idea to switch to DQ0 using only an option in the env in mind.

stheid, 2020-06-26:
Status quo:

reference in dq0 in the agent
state is in abc and returned by the env
actions to the env are also in abc

-> we just need to change the reference to abc and put it into the env.

stheid, 2020-06-26:
- An environment is a network.
- A network consists of components. Components are "loads" and "inverters".
- An inverter can take two roles: 'master', 'slave'.
- a master inverter has:
    - out: a voltage and current setpoint
    - out: a voltage and current measurement
    - in: gets modulation gain for DC-link-voltage 
    - the current setpoint can be 0.
    - one setpoint, measurement and action per phase.
	- all in all this means 3 inputs and 12 outputs
- a slave inverter is similar,
    - except it does not have a voltage setpoint
	- this means 3 inputs and 9 outputs
- a load has:
    - [right now] one current for each phase 
	     (in a stable net this is also a proximate power, as the voltage aproximates the setpoint voltage)
	- [later] power

stheid, 2020-06-26:
the network could be specified in the following yaml syntax.
commented lines are defaults that can be ommited alltogether.
Values are normalized to 0,1, therefore voltage setpoint is actually not needed.

```
net:
  inv1:
    id: lc1
    cls: MasterInverter
    u: [i1p1,...]    # names of the inputs
    v:
    - capacitor1.v
    - capacitor2.v    # <- alternative yaml array syntax
    - capacitor3.v
    i: [inductor1.i,...]
    # iref: [0,0,0]
    # vref: [1,0,0]
  inv2:
   ...
   ...
  load:
     id: rl1
     cls: Load
     i: [...]
```

The environment will provides a list of key words that can be map to the state output values. for example. if we do not specify the iref here in the configuration, the env will still return the abc references and they are named "inv1.iref1" ... However, yaml may not preserve order, therefore the agent needs to know how to map the values that are provided in the state as np.array to the values that it has to stick into its internal controllers. Right now we have the "obs_templates" for that and we probably only need to update them slightly to be able to work with the new structure of values.

stheid, 2020-06-30:
[image]
dotted lines indicate function calls that are done initially. the other lines are function calls in every step

stheid, 2020-06-30:
we need to bump python version again to 3.7 as i need the orderpreserving nature of python dictionaries https://stackoverflow.com/a/14959001/13310191

stheid, 2020-06-30:
Ideas to normalization

normalization should be done unit wise. all currents per inverter simmilarly and so on.
normalization should be customizable, beeing able to overwrite default logic
providing a normalization could be one with two integers a and b and normalized value is a+x/b
normalization could be provided by numexpr parsable strings

the parameter "norm" can be eighter an array of two integers or a string as numexpr. if two integers we will normalize all values with a+x/b.  otherwise with numexpr, where x must be the variable.

stheid, 2020-07-01:
### 1
- inverted Droop to slaveInverter
- finish normalization

### 2 @Webbah will conduct the following experiment:
- take the normalized environment with netfile.
- implement denormalization of the state
- use the same controllers to test that everything is docmented well enough and works propperly

### 3
- remove droop and pll from invertercontrollers

Webbah, 2020-07-14:
Checked net layer with is_normalized=False using two_inverter_static_droop_controller.py setup using saved results from the latter as validation: Different results in history, values out of range

stheid, 2020-07-16:
The bug is caused by an error that clearly shows a design flaw in handling the column names.

Its a little difficult to explain: Let the modelica class be M and the normalized Env be N: N will provide the agent with simmulation values, network related values and measurements previously passed from the agent. M only knows about simulation and measurement. But M will store the data in the recording class and M therefore also stores the variables names related to the recorded values. Hence only simulation and measurements. We are however a instance of N and therefore want behaviour that fully fits that.

I might need to add additional methods or extend the recorder class to cope with a more composite data recording. It is important to distinguish the variables names that are related to the simulation, and "the others"

wallscheid, 2020-07-17:
From discussion on 17th July: 
Pospone modifying the basic expert-controller examples to normalized mode later when RL-based agents are investigated to compare on same data interface.

stheid, 2020-07-20:
The issue should be more or less closed now. At least the code is done, although some parts will definetly need more polishing and documentation. However, i think it makes sense to give that another round when we add more parts to it as i think i need another view angle on the code in order to improve it conceptually.

stheid, 2020-07-24:
I think i will close the issue now and we might reopen issues to change parts of this code. Right now the Issue is not welldefined anymore (not clear when it is finished and what exactly should be solved.)

closed 2020-07-24 by stheid