upb-lea/openmodelica-microgrid-gym #48: Constraint Monitoring

wallscheid, 2020-07-02:
Currently we do not have a proper constraint monitoring class at all. In the provided examples, parts of the state space are used in order to manually calculate a performance metric which is considered to be safe if a policy performes better than a certain lower threshold value. This work for the specific SafeOpt application context, but is not very scalable.

The proposal is therefore to add a _ConstranintMonitoring_ class:
* Receives the entire state (maybe also most recent action) information and at each time step
* Processes the state/action information via arbitrary (user generated) constraint evaluation functions e.g. simple box constraints on single variables or more sophisticated nonlinear functions covering multiple states/actions.
* The class's output is a scalar between [0...1]:
  * 1 would be considered a hard constraint violation leading to an episode termination
  * 0 would reflect totally safe system behavior far away from any constraint problem
  * Feedbacks in between could be considered 'soft constraint' information i.e. we are somehow close to a limit/constraint. 
* This scalar feedback could be also used to implement a global barrier function which is aggregating the above's constraint feedback to shape the reward feedback (i.e. reward penaly for constraint feedbacks > 0).

Proposal could be added/integrated to issue #45 .

stheid, 2020-07-28:
is part of the Env class. Env calls it in every step.

can be configured by list of constraints with respect to variable names of the env.

The penalty could be returned in the 'info' {'penalty': .6}

stheid, 2020-07-28:
constraint monintorer can get
- a list of functions
- an aggregation function: (default: max)

validate return values to match the output domain (no clamping)

stheid, 2020-08-04:
I think this can more or less be implemented competely within this network and component classes and configured using the yaml configs.

stheid, 2020-08-05:
https://github.com/upb-lea/openmodelica-microgrid-gym/blob/36972d8786230d100a93b3c1569408c6d1b7f737/openmodelica_microgrid_gym/net/net.py#L217

Here we would need the implementation of the Grenzlastintegral. The init function could take the parameter and another function could be created to calculate the risk locally inside the component.

h-bode, 2020-10-08:
@stheid: seems like there is something broken. 

For example the two_inverter_static_droop_control.py returns following plot:

[image]

Same parameters as in the current develop branch, i even copied the fmu directly from there. must be somewhere in the data transfer, since the parameters are indentical

h-bode, 2020-10-09:
@stheid pytest does not work. When I noticed it, I checked out a fresh version of this branch.

7 Fails, 4x modelica.py and 3x runner.py

stheid, 2020-10-16:
The branch is now rebased onto develop and squashed. The changes are now super clean as most of them have been cherrypicked manually to develop before. With that major cleanup i guess it is doable again to finally tackle the actual task in this issue :D

stheid, 2020-11-27:
issue is basically solved and implemented however @Webbah wanted to have another look at the technical details, hence i assigned him the ticket.

Webbah, 2020-12-18:
looks awesome

closed 2020-12-18 by Webbah