upb-lea/openmodelica-microgrid-gym #50: Single Phase System Model

Webbah, 2020-07-09:


Webbah, 2020-07-15:
Define states and actions:

states x: f, U, ...?
- global (net-wide)
- component dependent

Webbah, 2020-07-15:
component power as state or input?
Due to dw=sum(P) $a$

Webbah, 2020-08-03:
$X_1 =50$

Webbah, 2020-08-03:
First example: 
* voltage considered as fixed
* two inverters connected via an inductor, resistive load between inv1 & ground.
 - leads to B = [B12, -B12; -B12, B12] and G = [G_load, 0; 0, 0] 

dtheta = f
df = sum(Pk)/(J*f)

While in Pk an offset can be considered and in f-ODE droop is inserted: 
`df = (p-droop_linear*(freqs-nomFreq))/(J*freqs)`

Result for droop inverter 1 = 0, inverter 2 = 1000 W/Hz and no P-offset: 

[image]

Webbah, 2020-08-03:
Unstable for longer simulations: 
[image]

[image]

* Change solver
* Calculate sign in matrix by hand to double-check due to it is working with -G (`p[k] += nomVolt * nomVolt * (-G[k][j]*np.cos(thetas[k] - thetas[j]) + \
                                         B[k][j]*np.sin(thetas[k] - thetas[j]))`)

h-bode, 2020-10-22:
Q-Droop and voltage implemented:

Observations: 

- Frequency at the load bus (f3, green) does not is not stable. Depending on the (p)-droop values, it slowly increases or decreases over time. 
```
droop_linear = np.array([7689, 1000, 0])     # W/Hz
q_droop_linear = np.array([12000, 3000, 0])
```
When the first value is below 7689 (does not matter how much), the frequency rises, when its 7689 or above, it drops.

[image]

[image]

continues even for longer simulations.  ignore the last value which goes to 0, this is just a programming issue.

[image]

- voltage of the load bus goes to 0. 

[image]

The other 2 voltages can be stabilized with the q-droop, but sinve it´s 0 for the load, it reaches 0.

I would have expected it to rise while adjusting (increasing) the load resistance, but this does not seem to have any effect on the resulting voltages?! Increasing it by the factor of 10 or 100 does nothing to it, while decreasing the line impedances results sometimes in numerical issues.

h-bode, 2020-10-22:
If anyone has a clue how to get a voltage at v3, which does not decrease that fast, feel free to share ideas :-)

h-bode, 2020-10-22:
Relevant parts of the code, basically equation 8 from Olivers paper: 

` for k in range(num_nodes):
        for j in range(num_nodes):  # l works, due to B is symmetric
            # Assume Voltage as constant
            p[k] += voltages[k] * voltages[j] * (-G[k][j]*np.cos(thetas[k] - thetas[j]) + \
                                         B[k][j]*np.sin(thetas[k] - thetas[j]))
            q[k] += voltages[k] * voltages[j] * (-G[k][j]*np.sin(thetas[k] - thetas[j]) + \
                                         B[k][j]*np.cos(thetas[k] - thetas[j]))`

Including equation 12 for the return, + an adjusted version for the voltage:
```
    df = (p-droop_linear*(freqs-nomFreq))/(J*freqs)
    dv = (q - q_droop_linear * (voltages - nomVolt)) / (J * voltages)
    dtheta = freqs * 2 * np.pi
```