upb-lea/openmodelica-microgrid-gym #53: Active loads using current sources

h-bode, 2020-07-17:
Under specific circumstances, modeling active loads by current sources seem to work.

[image]

Main Problems: 
- Unstable if an eigenfrequency of the LC filter is hit
- Resistors which simulate losses in the LC filter for damping are required
- Error while simulating in OpenModelica occurs for unknown reason, but does not effect the simulation process or the result: _division by zero in partial equation: sqrt(network_singleInverter22.activ_z.pll_ueff.add1.y)
at Time=0.000000
solver will try to handle that._

First results after tuning of the LC-filter: p = 1kW, q = 2kVAr

[image]

h-bode, 2020-07-17:
Same filter, but different load (p=2kW, q = 500VAr), totally different result!
[image]

h-bode, 2020-07-17:
Adding a butterworth lowpass at every phase current a,b,c, the previously bad results looks way better. The question is which influence a phase shift in this position has... 

[image]

wallscheid, 2020-07-17:
Phase shift would mix up active and reactive power flow. That should be monitored and corrected e.g. by an additional controller :)

h-bode, 2020-07-17:
Tuned Butterworth at the effective voltage in combination with a change of the "blackstart catcher" seem to solve the problem in OpenModelica. The resistor in the LC filter is still required, reducing it will increase the timespan oscillations occur in the signal

First try to control the inverters with the toolbox in python were not really successfull

closed 2020-08-07 by h-bode