upb-lea/openmodelica-microgrid-gym #55: JOSS Review Notes to Improve Stuff

wallscheid, 2020-07-24:
Feedback of katyhuff : 
- [ ] Add readme information for highlight OS support (SH)
- [ ] Classifiers in the setup.py accordingly (SH)
- [ ] Send thanks feedback on JOSS issue (SH) 
- [ ] Add DOIs to joss paper in markdown (HB)
- [ ] Add in-line comments to high-level tests (more information to new users on the test background) (SH)
- [ ] Add information to role of E. Hüllermeier (within review answer) plus add section 'individual contributions of the authors'. (HB) 
- [ ] Add Jarren to thanks section (HB)
- [ ] Clearification of the functionality scope (HB)
- [ ] Add link to whitepaper and refer to detailed example usage information there (small paragraph on scope of examples) (HB)
- [ ] Just go through comments of uploaded PDF and tune JOSS paper accordingly (HB)

stheid, 2020-07-24:
@h-bode  please use branch https://github.com/upb-lea/openmodelica-microgrid-gym/tree/feature/joss-katyhuff

h-bode, 2020-07-27:
Updated and pushed everything. 

@wallscheid , do you want to take an additional look?

@stheid since you made every communication in the past, are you going to present the new version to katy huff?

stheid, 2020-07-27:
Where did you do the changes? the branch i mentioned above does not have a single commit from you

stheid, 2020-07-27:
lgtm now #57 can be merged from my side

closed 2020-08-03 by wallscheid