upb-lea/openmodelica-microgrid-gym #58: Setting parameters via Python

h-bode, 2020-07-31:
There seems to be a bigger problem (bug?!) by setting parameters via PyFMI.

In my view, this are some problems either in PyFMI or in OpenModelica.

The whole issue in a minimal example is described in 

https://stackoverflow.com/questions/63191419/pyfmi-setting-parameters-sometimes-not-possible-model-depending

Added an additional reference to this as an issue at the PyFMI Github.

https://github.com/modelon-community/PyFMI/issues/48

h-bode, 2020-07-31:
Created also a trac for OpenModelica with the same content

https://trac.openmodelica.org/OpenModelica/ticket/6080#ticket

h-bode, 2020-08-20:
Finally, this issue seems to be solved!

The reason why it previously worked sometimes and sometimes not is still not clear for me, but with the following method, every parameterchange hat direct influence on the simulation result. 

At first, a basic initialisation needs to be done.

Example: 

```
    model.setup_experiment(start_time=Tstart)
    model.enter_initialization_mode()
    model.set('resistor.R', 1000)
    model.set('inductor.L', 1)
    model.exit_initialization_mode()
    time = Tstart
    model.enter_continuous_time_mode()
    x = model.continuous_states
```

```
During the runtime of the simulation, the initialisation mode needs to be reentered

    while time < Tend:
        n = n + 1
        if n == 200:
            model.enter_initialization_mode()
            model.set('resistor.R', 100)
            model.exit_initialization_mode()
            print(n)
        time = time + dt
        dx = model.get_derivatives()
```

Then, every parameter which is set has an impact on the simulation.

Changing the code in the modelica.py from line 318 to:
```
        if self.model_parameters:
            values = [(var, f(self.sim_time_interval[0])) for var, f in self.model_parameters.items()]
            # list of keys and list of values
            self.model.enter_initialization_mode()
            self.model.set(*zip(*values))
            self.model.exit_initialization_mode()
```

The problem should be fixed, but entering and leaving the initialisation mode takes some time. 

3 runs of the default two_inverter_static_droop_control without entering/leaving this mode in every step: 

Average time: 12.04
 With this modification: 
Average time: 12.67

If those lines are just run when there is a parameter change, we might speed up this a little, but the basic problem seems to be fixed!

stheid, 2020-08-20:
Thats very nice. We might be able to write a nice wrapper class for the pyFMI model to ease coding.

h-bode, 2020-08-21:
@stheid can you please write the wrapper class and put it (whenever it fits) in the develop? :-) 

If there are some problems with the wrapper, adding it like it is would be fine I guess, too.

Webbah, 2020-08-24:
Does not work in #59

wallscheid, 2020-08-24:
Great!
Please enquire with @h-bode  to investigate.

h-bode, 2020-08-24:
Yep, we discovered this issue this morning, when we were trying to set up Daniels Monte Carlo simulation.

Found some really interesting behaviour in different FMU´s. There is a workaround for Daniel, which enables him to du everything he likes to do, but not a final solution

Webbah, 2020-08-25:
Kind of conflict in #59 due to
 https://github.com/upb-lea/openmodelica-microgrid-gym/blob/26051df73743aee8584383f2bf82a674129e9f0f/openmodelica_microgrid_gym/env/pyfmi.py#L88

if commented: 
 - Possible to set parameters (L&R) in single_inverter_current_control_safe_opt.py 
**BUT**
 -  no fmu output in  single_inverter_voltage_current_control_safe_opt.py
      - means: the observations in env._simulate() are zero even if the actions are set correctly (Controller are working but env does not react):
[image]

if uncommented (**only change**: uncomment https://github.com/upb-lea/openmodelica-microgrid-gym/blob/26051df73743aee8584383f2bf82a674129e9f0f/openmodelica_microgrid_gym/env/pyfmi.py#L88 and https://github.com/upb-lea/openmodelica-microgrid-gym/blob/26051df73743aee8584383f2bf82a674129e9f0f/openmodelica_microgrid_gym/env/pyfmi.py#L90):

 - Not possible to set parameters (L&R) in single_inverter_current_control_safe_opt.py 
**BUT**
 -  fmu output in  single_inverter_voltage_current_control_safe_opt.py works fine

[image]

h-bode, 2020-08-25:
Thanks for bug report. I´ll check the FMU with my test programm, which gives easier information. From now, I have absolutely no clue why this happens, but maybe I´ll find something

wallscheid, 2020-08-25:
Is this strange behavior the revenge of FMPy not being considered by us any longer?

h-bode, 2020-08-26:
With the comand model.initialize() , it seems to work. Daniel does further testing. If he does not find problems, it will me merged together with the measurementbranch #59

closed 2020-08-26 by h-bode