upb-lea/openmodelica-microgrid-gym #59: Measurement

Webbah, 2020-07-31:


Webbah, 2020-07-31:
toDo: double check reward function
J is not in the range it should be  

[image]

wallscheid, 2020-08-03:
Ideas:
* Check/plot against reference values
* Check reward function calculation (bug?)
* Current sensor offset check
* Check if (parts) of the harmonics come from sampling during or shortly after switching instants of the IGBTs

JarrenLange, 2020-08-03:
Feedback from the hardware has been extended to include the direction reference values. 

Working on obtaining required hardware for current sensor verification.

For Sampling modification, changes to the hardware to choose when to sample and/or to change the internal carrier (forward,reverse, symmetric) are underway.

JarrenLange, 2020-08-04:
Using the setpoints from the FPGA hardware seems to solve most of the problems. Turns out calculated SPs (derived from the I_D setpoint and the phase angle) differed significantly.

Webbah, 2020-08-04:
Second try 1D for ki (kp = 0.01 1/A) current controller layout.

Resulting GP for ki_init = 20 1/(As), lengthscale = 5 /(As):

[image]

Initial currents: 
[image]

"Best" Result for ki = 11.7 /(As):
[image]

toDo:
* Inital ramp up not the best, tune kp ->2D problem!
* Safe theshold here choosen = 0 due to higher exploration and testing. But 0 would mean infinit error! Choose above 0 from now on

wallscheid, 2020-08-04:
great intermediate result! nice progress

quick question on the current waveforms: it seems that the three phase current is not idea, even in steady state (e.g. blue current is always too large for negative peak and green current is always too low for positive peak values). did you run an offset and gain compensation for all three current sensors before runing the experiment? Or do you have other ideas where the asymmetry / harmonics is coming from?

JarrenLange, 2020-08-04:
It seems to stem from an unbalanced filter inductance, especially in a short circuit and with low V_dc values. 

DQ0 control struggles with this. We do see the imbalanced currents in the verification probes. So I do not think it is an error with the measurement circuit.

Webbah, 2020-08-04:
Intermediate result in 2D (x->kp, y->ki; interrupted by ss connection error)

[image]

initial:
[image]

currently best one (green dot, first pic)
[image]

wallscheid, 2020-08-04:
Very nice result for 2D optimization. Here also the asymmetry is mostly gone -> interesting.

Webbah, 2020-08-04:
Second try, higher lentghscale (reaches unsafe point (kP=0 in plot below), but with less results more exploration...)

[image]

Yes, also first thought maybe tuning... but unbalanced L the case, too....

Webbah, 2020-08-06:
[image]
Comparison for same tuning of the current controller, showing only current through phase a at the moment

Webbah, 2020-08-07:
[image]
 
error(kP,kI) landscape from testbench

Webbah, 2020-08-10:
Plan:
Controller layout in simulation (Pt1 -> RL & Pt1 -> Inverter), 
Magnitude Optimum -> Kp&Ki_init
Estimate maximum lengthscale -> Use for measurement

- Voltage Controller -> same

- Apply loadsteps; wait for steady state -> full load -> half load

wallscheid, 2020-08-13:
Idea to make the simulation model becomer closer to reality:
add white noise to the current (and voltage) 'measurements' within the Python interface to the OM-FMU container.
mean = zero
Variance = take a snapshot time period from the test bench with actively swichting transistors controling a zero current / zero voltage. the measured signals can be then assumed to be the pure noise component and used to estimate the variance of the noise process.

JarrenLange, 2020-08-13:
Issue with sensor overcurrent is likely an underperforming op-amp.

Sensor uses a transimpedance amplifier to convert the current output (25mA nominal) from LEM LAH 25-NP (which has a 1:1000 conversion ratio). The op-amp (TL082) only has a 20mA maximum source/sink. 

Options:
 
1. Buy pin compatible op-amp with ~60mA source sink limit.
2. Modify circuit to let the LEM device terminate over the resistor and use the op-amp for general gain/filtering.
3. Use transistors in a push-pull combination on the output of the op-amp.

Investigating option 1 at the moment.

Webbah, 2020-08-17:
Noise measurement for current measurements from testbench: max(standard deviation) of 0.11 A detected

- [ ] Add measurement noise to env-step

At the moment len(obs)*std(noide_measurement) is taken. 
For voltage control that has to be adjusted

- [ ]  Assign different noise do different CVs

JarrenLange, 2020-08-17:
Adjusted the current sensors using option 2 as mentioned above. Different gains for system (3.6A/V to 6A/V).

But we can now have currents over 30A without non-linear behaviour 
[image]

Webbah, 2020-08-20:
**Current controller layout using safeopt - Testbench**
- Controller layout using magnitude optimum (measurement noise included)
- figured out maximum possible ('safe') lengthscale using simulation setting

[image]
(y-axis: lengthscale for kp, x-axis: lengthscale for ki; green -> safe; blue circle -> chosen point for measurement)

Result:
[image]

Initial controller tune:
[J_1.0_abcInductor_currents.pdf](https://github.com/upb-lea/openmodelica-microgrid-gym/files/5102607/J_1.0_abcInductor_currents.pdf)

[J_1.0_dq0Inductor_currents.pdf](https://github.com/upb-lea/openmodelica-microgrid-gym/files/5102609/J_1.0_dq0Inductor_currents.pdf)

**Best result**
Performance increase by 11.58 %
But more noise on it, run for longer?
Weight the initial error less or the "later" error  ('noise' around the setpoint) higher?
So slower step response but less noise?  
[J_1.1158069816015708_abcInductor_currents.pdf](https://github.com/upb-lea/openmodelica-microgrid-gym/files/5102619/J_1.1158069816015708_abcInductor_currents.pdf)

[J_1.1158069816015708_dq0Inductor_currents.pdf](https://github.com/upb-lea/openmodelica-microgrid-gym/files/5102620/J_1.1158069816015708_dq0Inductor_currents.pdf)

Webbah, 2020-08-20:
First try with voltage controller running on open circuit using analytical current controller layout results.

[image]
 
Initial result (without analytical layout): 
(Performance in the title)

[image]

Best result after 100 epsiodes:
[image]

wallscheid, 2020-08-20:
Are the following stuff results from the test bench or simulation? Was not 100 % sure from your text input.

> [image]
> 
> [image]
> 
> **Best result**
> Kpi = [0.9626728565572376, 24.507337238803796] (green circle in the agent plot above)
> Performance increase by 11.58 %
> But more noise on it, run for longer?
> Weight the initial error less or the "later" error ('noise' around the setpoint) higher?
> So slower step response but less noise?
> 
> [image]
> 
> [image]

Regarding the noise in steady-state: Did you use the mean root error (MRE) as the cost criterion (as in the preprint paper) or something different? Normaly the MRE is quite a suitable cost function to ensure nice behavior in steady-state since smaller control errors are not squeezed-together compared to the standard MSE.
Nevertheless, with respect to the results the noise amount is increased, yes, but with the initial controller parameters there was also some significant harmonics on the current which are suppressed with the optimal tuning design.

wallscheid, 2020-08-20:
Additional question regarding the current controller results: How close are not the behavior from the simulation and the experimental optimization? Is the performance landscape and the optimal found controller gains both similar in simulation and experiment?

Webbah, 2020-08-21:
**This is simulation**
[image]

[image]

Initial episode: J = 1
[image]
[image]

Best episode: J = 1.25
[image]
[image]

JarrenLange, 2020-08-24:
I have checked the DC-Link behaviour of the system during the open-circuit and short circuit tests.
Here are the waveforms
Under short circuit (first 100ms of DC bus noise is with no modulation, performing a current nulling test).
[image]

Under open circuit
[image]

wallscheid, 2020-08-24:
I do not get the voltage measurement. I would expect to see the average DC-link voltage (at around 60 V?) and then some ripple and a possible dip (when starting the current control) on top. Assuming that above the orange signal is the DC-bus, why it is around 0 V?

JarrenLange, 2020-08-24:
Orange is the VDC bus (AC Coupled), the bottom display is the currents or voltages.

Note, this measurement was taken using the custom voltage sensors. I will test it soon with an independant differential probe.

JarrenLange, 2020-08-24:
[image]

Repeated short circuit test, but with 2 sensors for the VDC (Given here in absolute units).

Webbah, 2020-12-18:
results in paper, branch mereg to develop

closed 2020-12-18 by Webbah