upb-lea/openmodelica-microgrid-gym #61: Publishing OpenModelica Library

h-bode, 2020-08-07:
At https://www.modelica.org/libraries , several user defined libraries are published. 

Contacted the website which requirements exist to get listed there, and if we should provide any additional information, pictures etc.

Until now, no response.

wallscheid, 2020-08-20:
Still no response? 
If yes, please send a reminder.

h-bode, 2020-08-20:
After getting no response from the general contact site, i wrote on wednesday directly to Michael Tiller, the "owner" of this library project, but he has not answered yet

h-bode, 2020-09-04:
Published a first version of the library on my private account.

A structure like this decreases the time for modelling DRASTICALLY, since not everything is stuffed to only one huge file. 

https://github.com/h-bode/OpenModelica_Microgrids

I have send this to Michael Tiller and Sjölund for an additional feedback, if this library can be published on its own.

If everyone is fine with it (and if we have a final name for it) I would like to implement this form in the OMG toolbox as well.

I think that not much would change for the python code, only the name of the FMU will be different.

h-bode, 2020-09-05:
According to sjoelund, we might advertise the toolbox itself at https://modelica.org/tools, too :)

h-bode, 2020-09-08:
https://www.modelica.org/licenses/ModelicaLicense2

I think we could use this license for the library, seems to include everything we want, it is recommended by the the OpenModelica board, and other universities use it to publish their libraries, too.

h-bode, 2020-09-09:
Published at the Modelica Libraries. 

The OMG Toolbox itself will be published at: https://modelica.org/tools soon.

The description is taken automatically from description in the GitHub, which is the part below the _about_ section. So we can update it without contacting the Modelica guys.

closed 2020-10-20 by h-bode