upb-lea/openmodelica-microgrid-gym #62: Getting ODE System out of OpenModelica

h-bode, 2020-08-07:
Since there are lots of bugs and problems with using PyFMI, extracting the ODEs directly from OpenModelica might be a way. 

We don´t need the main benefit from FMI liek connecting ODEs from different platforms, so if it is possible to extract the ODE system, we would still have the advantage of the nice GUI of OpenModelica but hopefully without the bugs and problems of the unsupported PyFMI stuff.

h-bode, 2020-08-10:
Answer from the Modelica-Forum (one of the OpenModelica developers from sweden): 

> There is no such output at the moment, and you won't ever be able to get a pure ODE system out of it since Modelica contains hybrid parts, the clocked system, and non-linear systems, dynamic state selection, etc. But for some models you might be able to get an ODE system out of OpenModelica. If you write an exporter for it or perhaps parse the info.json file to get the equations you want (for even more limited models).

h-bode, 2020-08-10:
In the FMU, the ODE system is pretty much spreaded over everything, too.

Most of the equations seem to be in the file: 

grid.network/source/OMCppgrid_network.cpp

But the initialisation, limits, breaks, jacobian-calculations etc are somewhere spreaded all over.
And we don´t know exactly where our problems occur. Is it really PyFMI, or does the problems occour in the FMU generation?

stheid, 2020-08-10:
Is there any alternative to PyFMI such that we can try to check the FMU without PyFMI?

h-bode, 2020-08-10:
There is an alternative, FMPy, which I am currently testing.

In opposite to PyFMI, it is under permanent development. I had it on the scope already in November, but to this time, it was not compatible to OpenModelica. In June, OM fixed some of the issues, so in theory, it should be compatible now (with the nightly build of OMEdit).

wallscheid, 2020-08-20:
@h-bode : Would you agree that the approach getting the ODE description directly from OM in order to integrate it afterwards within Python is not feasable?! From the above summary and personal discussions it seems not fruitful. 

If you agree, please close the issue.

h-bode, 2020-08-20:
Yes, totally agree. But since the main issue with PyFMI is solved, it is not to bad

closed 2020-08-20 by h-bode