upb-lea/openmodelica-microgrid-gym #65: OMG support for Macs

h-bode, 2020-08-19:
Since installing and using OpenModelica is a hell fpr Macs and no one of us owns one, i would suggest to drop the idea of supporting macs. 

Manually installing different programs with root access after the warning that there is no uninstaller for this programs, registering on additional sites and the strong recommendation of the OpenModelica guys to use a Linux VM instead prevented me from continuing on someone else´s private MacBook.

(https://openmodelica.org/download/download-mac)

A users OpenModelica version in theory is only needed for generating user defined grids. A limited use would be possible for Mac users if we could provide those FMUs.
But in my opinion, it would be smarter to follow the OpenModelica guys and suggest that Mac users should run everything in a Linux VM. Clean OpenModelica installation, clean PyCharm installation without pip problems (couldn´t install 4 packages (GPy, Gym, SafeOpt and something with _iter via pip, the first two only via condaforge)), and no problem for us to provide the FMUs. 

Or are there any other limitations I havn´t thought about?

wallscheid, 2020-08-20:
I believe when the OM guys really recommend using a Linux VM for Mac users, then we also should follow this path. Thanks @h-bode  for investigating.

@stheid : Could you (together with h-bode) please extend / modify the documentation and installation notes accordingly?

h-bode, 2020-08-21:
I will write something and show it to Stefan

stheid, 2020-08-21:
if you'll write the documentation, i'll assign it to you and than you assign me to the PullRequest and i will finish it up.

wallscheid, 2020-08-21:
Please also add a remark to the JOSS paper regarding the MacOS installation constraints.