upb-lea/openmodelica-microgrid-gym #74: Datapath for saves OS dependend

h-bode, 2020-08-26:
In 

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/measurment/examples/single_inverter_voltage_current_control_safe_opt.py#L178

the data path is OS-dependend. To be used in Windows, it needs to look like
```

    def xylables_v_abc(fig):
        ax = fig.gca()
        ax.set_xlabel(r'$t\,/\,\mathrm{s}$')
        ax.set_ylabel('$v_{\mathrm{abc}}\,/\,\mathrm{V}$')
        ax.grid(which='both')
        time = strftime("%Y_%m_%d_%H_%M_%S", gmtime())
        fig.savefig(save_folder + '\Abc_voltage' + time + '.pdf')

```

while in Linux, following is the way to go: 

```
    def xylables_v_abc(fig):
        ax = fig.gca()
        ax.set_xlabel(r'$t\,/\,\mathrm{s}$')
        ax.set_ylabel('$v_{\mathrm{abc}}\,/\,\mathrm{V}$')
        ax.grid(which='both')
        time = strftime("%Y-%m-%d %H:%M:%S", gmtime())
        fig.savefig(save_folder + '/abc_voltage' + time + '.pdf')
```

If we want to enable the possibility to generate this .pdf images, we need to adjust it.

Maybe with an if-case, and semi-redundant code (ugly, i know Stefan, feel free to suggest something nicer ;D )

stheid, 2020-08-27:
`os.path.join(savefolder, f'abc_voltage{time}.pdf')`

h-bode, 2020-08-27:
Thanks Stefan!

Works for Windows if you use following time format:

        time =  strftime("%Y-%m-%d_%H_%M_%S", gmtime())
        os.path.join(savefolder, f'abc_voltage{time}.pdf')

@Webbah , can you close this issue after using this in the measurement branch? I think an implementation together with this is sufficient.

closed 2020-08-28 by stheid