upb-lea/openmodelica-microgrid-gym #79: Change of the OpenModelica Library

h-bode, 2020-09-07:
I would like to change the OpenModelica Library to the structured one in my GitHub (see  #61)

To make sure everything runs and avoid problems after merging the measurement branch to the develop, I rebuild every model Daniel used in the measurement branch, all other affected files, and test it. When everything is done, I would suggest that I create a Pull request against the measurement branch and ask @Webbah to verify it.

h-bode, 2020-09-07:
FInished the implementation and adaption to the measurement branch. 

Had to change some of the names and datapath. The folder previously called "fmu" needs to have the same name as the library, "OpenModelica_Microgrids". 

The model_path in in toolbox became a little longer, its now for example:

model_path='../OpenModelica_Microgrids/OpenModelica_Microgrids.Grids.NetworkSingleInverter.fmu',

But everythink except the tests seem to work. I fixed the path to the test.fmu in pytest. The other failing tests seem for me to be realted to other adjustments like the added measurement noise. 

The branch with the new library is pushed. 

Tomorrow morning, I will check if the tests run in a modified version of the dev, to make sure, I didnt break something in the tests.

h-bode, 2020-09-09:
Everything seems to work from my side. 

All models from the measurement branch are already included (as .mo file, not as FMU). 

Waiting for Daniel to confirm, afterwards this issue can be closed.

Webbah, 2020-09-16:
Merged in dev

closed 2020-09-16 by Webbah