upb-lea/openmodelica-microgrid-gym #82: Switching losses (inverter)

h-bode, 2020-09-16:
While editing our first paper, the topic with the switching losses of the inverter came back in my mind. 

I think it is difficult to implement them in OpenModelica. In theory, they could be handled as an additional counter-variable in python. This would have just the effect of some measurement information without any "real" loss effects in the simulation. 

But we could at least explain some deviations between the model and the real testbench with it. 

Would it be worth to be implemented (in some future milestone :D ) in the toolbox, or is this topic not interesting enough, at least in a way it would be possible to be implemented?

wallscheid, 2020-09-16:
I believe it would be very interesting if we extend the scope of our investigations towards finite set control schemes, i.e., controllers with discrete actions directly switchting the transistors on/off. In this case, loss minimization in terms of an optimized modulation scheme / pulse pattern is a very typical research focus.

h-bode, 2020-09-16:
I totally agree, but I am not really sure what the best way to implement it. 

> I believe it would be very interesting if we extend the scope of our investigations towards finite set control schemes, i.e., controllers with discrete actions directly switchting the transistors on/off. 

Just assuming equivalent models based on the simulations in the python environment, or do you think that it might be possible to model in on OpenModelica? There is a semiconductor-toolbox, but behind the surface, the blocks just work with variations of a resistor value (very high and low).

One possibility in python might be writing a function for the losses depending on voltage, current, switching frequency(...) and simply add it with a factor to the cost function of the optimization?

wallscheid, 2020-09-16:
I believe a loss model for standard components (transistors, diodes, inductors, capacitors) can be easily integrated within the Python part of toolbox using the data generated by the OM simulation in a post-processing fasion. An integration within OM should not be necessary. 

For example, having the load current and the DC-link voltage during a switch on or off event available for a transistor will be enough to in order to extract an equivalent power/energy loss value from data sheet information of a given transistor type.