upb-lea/openmodelica-microgrid-gym #84: SafeOpt agent plots not show up any more

Webbah, 2020-09-17:
https://github.com/upb-lea/openmodelica-microgrid-gym/blob/4d9ec1489b3ebe271ffaea7b40b9e9198f4e1d67/openmodelica_microgrid_gym/agents/safeopt.py#L166

Since the optimizer increases in every step only the first plot is showing up.

stheid, 2020-09-17:
Oh, we thought `x.size` was the dimension, but its actually the number of dataelements?

h-bode, 2020-09-17:
Yep, this size increases every run by the number of dimensions. The easiest way to "bugfix" was multiplying the 3 with the iteration we are currently in, or has someone a better idea?

h-bode, 2020-09-17:
In theory this might work, but if I see this right, the agent does not know in which episode it is. And transferring the episode number just for this purpose through everything is not a really nice solution

stheid, 2020-09-17:
i assigned it to you, to validate the PR and close the issue if you are convinced of the solution.

Webbah, 2020-09-18:
Works after testing, thanks
all tests run through

closed 2020-09-18 by Webbah