upb-lea/openmodelica-microgrid-gym #87: IEEE 39 bus system

stheid, 2020-09-24:
I have no idea how this relates to controlling tasks, but it seems to be a fairly standard problem.

https://icseg.iti.illinois.edu/ieee-39-bus-system/
https://github.com/RLGC-Project/RLGC#customize-the-grid-environment-for-training-and-testing

I am not sure if this is worth implementing to give our work more comparability or if its completele unrelated to our microgrid controlling.

wallscheid, 2020-09-24:
First, thanks @stheid  for diging out this toolbox.

Here, we have to differentiate two things:
For the bus system layout (e.g. IEEE 39 bus) some pre-defined example topologies are existing in the literature. However, these are just providing examples on the connection structure within a local grid. No information regarding the specific components and their electrical power behavior within this grid is given. Hence, with such a bus system layout you cannot perform any system analysis or control-oriented simulation before not filling up the simulation framework with more “life”.

The mentioned toolbox is just using such a bus system layout as the basis for a simulation within a paper. However, the toolbox is poorly documented and specially it is using some 10-years old (not maintained) simulation toolbox which I have never heard of. Likewise, the model is not well documented within their referenced paper and actually many realization details of their environment model are completely in the dark. Hence, I find it more than irritating that the paper was already cited 21 times….

wallscheid, 2020-09-24:
The bottom line from my point of view is: the paper might be considered to be mentioned within a literature review for future papers from our group, but I believe it is not worth diging into their toolbox.

stheid, 2020-09-25:
We agreed that the toolbox is not directly relevant to us, however, we might want to cite and compare it in later publications.

Additionally if the swing equation simulation is available we might think of implementing one of those IEEE bus systems to provide a comparable problem setting.

closed 2020-09-25 by stheid