upb-lea/openmodelica-microgrid-gym #96: Parametervariation has no effect to simulation

h-bode, 2020-10-16:
Again, the problem with the parametervariation occured. 

Might be reasoned by the change from OpenModelica 1.16 to the nightly 1.17.

Featurebranch created to test (in windows) what the reason is.

h-bode, 2020-10-16:
Seems to be somehow connected to the new OpenModelica package structure...

Absolutely no clue what and how changed, since the names (and pathes) to the variables are completly identical.

The only thing which is different is that the ODEs in the toolbox structure are in different folders. If this is the case, we might need to go the step back to the old structure, which would be somehow......saaaaaaaad :-(

h-bode, 2020-11-03:
After some additional testing, Andreas closed this issue today. 

He checked both FMUs with an analysis tool, which stated for both FMUs, that for example, rl1.resistor1.R is a fixed value, which can´t be changed at all. For his understanding, it is because of a dependency of this value with another variable, but it´s unknown which one it is. 

The interesting point is that the analysis showed this problem for both FMUs, and according to his understanding, it shouldn´t work in both cases. He can´t explain why the old version is running without any problems. He expected errors/warnings from PyFMI instead, which do not occur.

Concluding, he couldn´t find errors in the FMU creation process itself, and stated that the error should be somewhere at PyFMI. 
So the issue is closed for him.

To be honest, I am not completely sure that he is right, but I don´t know what to say against his statement.

wallscheid, 2020-11-03:
Go next step: make PyFMI great again ;)

h-bode, 2021-01-25:
Since there is no further progress and no idea how to solve this issue, the issue needs to be closed. 

Will be reopened when some progress is therer

closed 2021-01-25 by h-bode