upb-lea/openmodelica-microgrid-gym #103: Add feedforward term to controller class

Webbah, 2020-11-16:
Add a feedforward term to output in https://github.com/upb-lea/openmodelica-microgrid-gym/blob/ba63ebb6d14fdb2365186eb8c7a8c8183503b505/openmodelica_microgrid_gym/aux_ctl/pi_controllers.py#L44
and https://github.com/upb-lea/openmodelica-microgrid-gym/blob/ba63ebb6d14fdb2365186eb8c7a8c8183503b505/openmodelica_microgrid_gym/aux_ctl/pi_controllers.py#L45
so it is considered in the anti-windup:
`clipped = np.clip(output+FF, *self._params.limits)`
`self.windup_compensation = (output+FF - clipped) * self._params.kB`

- changes tests?

stheid, 2020-11-16:
we should make sure, no-one confuses this with neural networks (feed forward). maybe by adding clarifying comments in the code

Webbah, 2020-12-18:
done in #59 merged to dev

closed 2020-12-18 by Webbah