upb-lea/openmodelica-microgrid-gym #109: Normalized env history

Webbah, 2020-12-09:
It seems like currently the normalized env outputs are stored in the history if `env.is_normaized == True`.
That leads to 'wrong' env plots showing the normalized values but not the real-world (simulated) measured:
[image]

Values should be [15,-15] A.

Re-normalize before stored to hist?

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/0ae67d388bb9f6c3e92af18641fe766bd99fe7bf/openmodelica_microgrid_gym/env/modelica.py#L318-L320

stheid, 2020-12-09:
The hotfix would obvoiusly be:

```python
outputs = self.net.augment(self._state, self.is_normalized) 
outputs = np.hstack([outputs, self.measure(outputs)]) 
raw = self.net.augment(self._state, False) 
raw = np.hstack([raw, self.measure(outputs)]) 
self.history.append(raw) 
```

another way would be to denormalize the data on rendering. I am not sure if that is all that elegant however.

I think we should think critically what we actually want to remember. i think remembering the unnormalized data is ok as long we make sure to return the normalized data. Hence the elegant solution would be along the lines of the first hotfix (maybe a little more DRY though).

Webbah, 2020-12-10:
Hotfix does result in a 100 Hz current.
- suggestion is that in the hotfix-denormalization the oscillator is executed

commit 10da62cf (2020-12-18): fix #109
closed 2020-12-18 by stheid