upb-lea/openmodelica-microgrid-gym #113: Bugfixing in the master

h-bode, 2021-01-08:
Some bugs are still left in the master (and dev, too).

-  Undersampling is not removed in every example. Please remove it in the examples "two_inverter_droop_safe_opt" and "single_inverter_voltage_current_control_safe_opt"

- Tn the example "two_inverter_static_droop_control", line 46: it t < .02 needs to be changed to t < .2. Documentation which is in line 40 can be moved closer to the code in line 46

- test_runner.py seems to fail depending on the platform. There are issues with the "test_main_paramchange"  @JarrenLange or @wallscheid, can you try to run the pytest on the master a windows? 

Originally, thetollerance of this test is 5e-3, which throws errors on my machine, but travis and linux PCs seem to be abled to handle it. setting this value to 5e-2 works for me.

stheid, 2021-01-08:
the first two issues are solved and pushed to develop. I am not sure if it makes sense to relax the tolerances as ultimately travis is the master of test.

stheid, 2021-01-11:
So the strange test results are hopefully handled with #115

closed 2021-01-11 by stheid