upb-lea/openmodelica-microgrid-gym #116: change reference values using callback?

Webbah, 2021-01-13:
Till now the reference values are changed in model_params as "side-function" while e.g. a model_parameter L is changed like

`        def reference_step(t):

            if t >= .05:
                i_ref[:] = i_ref2
            else:
                i_ref[:] = i_ref1

            return partial(l_load.give_value, n=2)(t)`

and used in the env definition as follows:
`model_params={'lc.resistor1.R': partial(r_load.give_value, n=0),
                                     'lc.inductor3.L': reference_step},`

Possible to  change the reference values without including it into a change of the pysical model parameters (L,R,C)?

stheid, 2021-01-13:
seems related to #99 

Maybe the issue can be solved by allowing general callback functions that are executed in the context of the `net.Components` or the `net.Network` class during before their `augment` function

Webbah, 2021-01-14:
In the RL-Example we will directly focus on voltage control, therefore the cascaded (traditional PIPI) structure is not needed.
- i_meas and v_meas are given to the agent. 
- v_ref (and f_ref) should be fixed?
- any reason to keep extra (VARIABLE) i_ref?

Webbah, 2021-01-18:
In the RL-Example we will directly focus on voltage control, therefore the cascaded (traditional PIPI) structure is not needed.
- i_meas and v_meas are given to the agent. 
@wallscheid :
- v_ref (and f_ref) should be fixed?
- any reason to keep extra (VARIABLE) i_ref?

stheid, 2021-01-28:
We discussed that it would be nice to have but not nessecary. Hence some time will be spent to try to implement it and if not successful document here on how it could be implemented quick and dirty here.

closed 2021-02-09 by Webbah