upb-lea/openmodelica-microgrid-gym #117: Add Stochastic processes to the Examples

stheid, 2021-01-14:
Currently load reference trajectories are very simple, but perhaps it makes sense to add one or two examples on how to create more random references to make the user aware how this can be implemented if needed.

https://pypi.org/project/stochastic/

stheid, 2021-01-15:
@wallscheid The wrapper is implemented, now we only need to configure a example with good default values that show how to use it.

wallscheid, 2021-01-15:
Assuming that the wrapper itself outputes normalized values, I would suggest to denormalize them with respect to the load configuration such that everything between full load and (nearly) no load is possible. Hence, for a given current limitation of an inverter we (i.e. @Webbah ) can calculate suitable load ranges which can be used to map the de-normalization to.

closed 2021-01-18 by stheid