upb-lea/openmodelica-microgrid-gym #121: Introduction Videos

h-bode, 2021-01-20:
A suggestion of the first intro video for the toolbox. The idea was just to show what is possible, like a teaser.

    https://youtu.be/UxyTr2p-ClM

KSS gave me the feedback that everything was way to fast, for everything i presented I should use twice the time.

Everyone with the same opinion?

stheid, 2021-01-20:
Agree, definetly extremely fast, but i would check how it looks if you just slow down the video without re-recording everything

h-bode, 2021-01-20:
Just reducing the audio speed sounds horrible, i tried it while using 0.75x speed directly in youtube.

wallscheid, 2021-01-20:
Holla the woodfairy, way too fast. And I agree that reducing the speed within the video is not an option since the sound get suboptimal.

I would suggest re-recording the video with a general speed decrease of 50 %. Also some more introductory words, i.e., what is the purpose/background of OMG and the video itself (both just briefly)?!

h-bode, 2021-01-22:
https://youtu.be/8cFKwC0GC28

This version should be way better now

h-bode, 2021-01-24:
https://youtu.be/19mBu6XSC_8

Now with some more explaining text during the examples at the end

h-bode, 2021-02-01:
Second video about the two inverter static droop control:

https://www.youtube.com/watch?v=Ef8NffFxtew