upb-lea/openmodelica-microgrid-gym #122: Test issues windows

h-bode, 2021-01-20:
Running Pytest with windows generates an error. The second tests of the runner causes an assertion error. 
Running the same file in Linux doesn´t cause problems.

Idea was that the fmu might be different. The test.fmu which is used for linux is the grid.network from april 20th, 2020.

I restored the grid.mo of this date, created a new test.fmu, and ran it, same error as the original one. Daniel did the same on his Linux, no errors. 

And now the interesting part: using the actual grid.mo from the dev to create the test.fmu causes 5 errors (runner and modelica) ob windows, doing the same on Daniels linux machine doesn´t cause an error at all.

The differences between the OS seem to be larger than we expected...

stheid, 2021-01-25:
related to

#115
#113

wallscheid, 2021-01-25:
additional headline in readme to highlight how to execute the pytests plus comment on windows-related issue (with link)