upb-lea/openmodelica-microgrid-gym #131: Align max_episode_steps - agent & env

Webbah, 2021-02-19:
https://github.com/upb-lea/openmodelica-microgrid-gym/blob/703c67af5847e5032718394102e3f08fe81c9039/openmodelica_microgrid_gym/env/modelica.py#L114-L115

Because of python counting zero based we run with the upper definition `max_episode_steps + 1`.

The training algorithm using e.g. stablebaseline3 count till `max_train_steps - 1` leading code like:
`model.learn(total_timesteps=1000, callback=callback)` and   `env.max_episode_steps = 1000` being rollout twice.

My suggenstion would be to change the definition in our env.

Any suggestions/oppinions/other ideas?

stheid, 2021-02-19:
i think what would be better is to have 1 state from the reset and then 10 actions and 10 env steps. i think you can simply change the start or endtime of simulation accordingly such that the calculation is correct.

Webbah, 2021-02-19:
Solved by increasing timesteps by 1 in env.init

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/168b208a685cc1df12036b062403f3b950580862/openmodelica_microgrid_gym/env/modelica.py#L116-L117

closed 2021-02-19 by Webbah