upb-lea/openmodelica-microgrid-gym #133: droopParams in components.py

Webbah, 2021-04-07:
In the Component class reference values are calculated to make them available to e.g. an RL-agent as input from the "environment".
To calculate the reference values the wanted frequency is needed to calculate the phase needed for ref_abc.

Currently we have defined the droop-parameters in the component class, too:

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/1036dfd80d2f3ccb54c061d33f9332cbe99cd9bd/openmodelica_microgrid_gym/net/components.py#L125-L126

https://github.com/upb-lea/openmodelica-microgrid-gym/blob/1036dfd80d2f3ccb54c061d33f9332cbe99cd9bd/openmodelica_microgrid_gym/net/components.py#L97-L98

Problem:
- Twice droop params are defined
1. main script (static or as mutuable float to be optimized)
2. as default values in the component.py

If the droop values differ, that leads to different values in the (doubled?) reference values in the env history:
(master.SPVa: internal value from controller calulation;
inverter1.v_ref.0: from env.components logged ref value)

[image]

- what happens if we optimize the droop controller value?
[image]

the droop-params in the net will not be changed!
1. Do we need that?
2. is the "env"-ref value needed on the control-part on the agent side?
Or is it only needed in case of feeding an RL-agent?

Solution-suggestion:
Delete all droop-params in the components.py and make clear, that droop has to be considered externally.
@wallscheid Do we have a problem in that case if we want to "learn droop-control" with RL-agents?

wallscheid, 2021-04-07:
If an agent is still able to alter the droop parameters acting on the control loops, I do not see any issue.

Webbah, 2021-05-18:
Compromise: Left droop-params in the component class but set per default to zeros.
In that case it is possible to use them but per default they have no influence (like MasterInverter per definition calculates f(P) and does not reduce f if pDroopParams are zero).

closed 2021-05-18 by Webbah