upb-lea/openmodelica-microgrid-gym #139: Example doesn't work right out of the box.

JvPost, 2024-07-02:
Im trying to run the examples single_inverter_current_control_safe_opt.py and single_inverter_voltage_current_control_safe_opt.py.

Both have an issue with the reward function. 
Specifically, the line `phase = data[idx[1]]  # phase from the master controller needed for transformation`  has an error. It's the fourth line in the reward function. 
The data passed to the reward function has not enough indeces to follow the pattern performed in the set_idx method.

I'm running on wsl in python=3.8.18. I've installed the packages in the requirements.txt and the omg package.

stheid, 2024-07-02:
Do the simpler examples work for you? Unfortunately, i have no installed version of the code any more.

What exactly are you trying to achieve.

Webbah, 2024-07-02:
Can you send a screenshot of how the error and data and idx look like?

On my end it is working and looks like that:
[image]

Kalash-byte, 2024-09-02:
[image]I also got the same error but after minor tweaking i was able to resolve the error. But next problem i am facing is this.